asterics/esp32_mouse_keyboard #2: Media keys

untitledlt, 2018-01-04:
Is it possible to send Media keystrokes like Play/Pause/Next?
Could you please provide an example?
And if i get it right, UART2 is not enabled on pins, right? How can I enable it?

benjaminaigner, 2018-01-05:
I didn't try Mediakeys yet. Please consider this as a very early project.
We are currently changing this whole project to provide the full FABI / FLipMouse firmware (see asterics/FABI and asterics/FLipMouse).

But the BLE part will be the same, so it will be supported some time. If you want to try:
send a keycode via esp_hidd_send_keyboard_value and use "KEY_MEDIA_PLAY" defines (keylayouts.h) as keycode values. And please give feedback if it works :-) 

Regarding UART: please have a look at the esp-idf examples, there is a detailed example on how to activate UART and corresponding pins.

benjaminaigner, 2018-01-13:
@untitledlt 

Hi, did you try the mediakeys?

untitledlt, 2018-01-13:
@benjaminaigner nope, not yet

chegewara, 2018-05-04:
I think this can be done with extra reportMap and input characteristic. You can use Consumer pare to do it. Chapter 15 in this document:

http://www.usb.org/developers/hidpage/Hut1_12v2.pdf

chegewara, 2018-05-08:
http://www.usb.org/developers/hidpage/HUTRR42c.pdf

> Example Consumer Remote Control Report Descriptor and Report
> 0x05, 0x0C,      //  USAGE_PAGE(CONSUMER_DEVICES) 
> 0x09, 0x01,      //  USAGE(REMOTE_CONTROL) 
> 0xA1, 0x01,      //  COLLECTION(Application) 
> 0x85, REPORT_ID_HOTKEY,  //    REPORT_ID(REPORT_ID_HOTKEY) 
> 0x95, 0x01,      //    REPORT_COUNT(1) 
> 0x75, 0x10,      //    REPORT_SIZE(16) 
> 0x15, 0x00,      //    LOGICAL_MINIMUM(0) 
> 0x26, 0xFF, 0x03,     //    LOGICAL_MAXIMUM(0x3FF) 
> 0x19, 0x00,      //    USAGE_MINIMUM(0) 
> 0x2A, 0xFF, 0x03,     //    USAGE_MAXIMUM(0x3FF) 
> 0x81, 0x00,      //    INPUT(Data_Arr_Ab) 
> 0xC0        //  END_COLLECTION

benjaminaigner, 2020-06-25:
Dear @MrSwiss ,
I've updated this repo in this branch https://github.com/asterics/esp32_mouse_keyboard/tree/bumpEspressifExample
If you want to give it another try,
you can use the function `esp_hidd_send_consumer_value` together with the `HID_CONSUMER_PLAY` macros for the different media keys.

If you are no longer interested (I understand, this issue was opened a long time ago), please close this issue.

benjaminaigner, 2020-07-20:
Media keys are working now, I've added a test code for mute/vol up/vol down in this repo (with 6deb4c5).

closed 2020-07-20 by benjaminaigner