asterics/esp32_mouse_keyboard #3: How to compile w/ platformio of w/out

Ericvf, 2018-01-12:
Hi there,

This looks very promising. I've used ESP32 on Windows with PlatformIO before but I don't know how to compile these files. I'm used to using the Arduino framework but this uses ESP-IDF correct? I'm trying to build this with PIO but it complains about esp_bt.h. I'm not sure how to resolve this, it's quite confusing. PIO identifies boards, platforms, frameworks and libraries. ESP-IDF qualifies as a framework and cannot easily be updated to the latest version I believe. 

I'm not seeking on guidance on how to do this with PIO perse. Can you provide some guidance on how to get up and running on Windows 10? Where do I find the toolchain and `make monitor console`? If you can just point me to a good resource that would be fine! 

In any case, I was glad to find this repo. This would be ideal as a bluetooth/hid proxy for environments that don't support bluetooth (don't get me started on my employer). 

Thanks

ChrisVeigl, 2018-01-13:
Hi,
we use the espressiv IDF toolchain under Linux: 
https://github.com/espressif/esp-idf

There are also setup instructions for windows (see readme.md).

benjaminaigner, 2018-01-13:
@Ericvf 

Hi, thanks for the interest on this project :-).
If it complains on esp_bt.h, it might be possible you are using an outdated version of the esp-idf.
A few weeks ago, bt.h was defined as deprecated and the suggestion was including esp_bt.h.

If you cannot update the esp-idf, please adjust this header file from esp_bt.h to bt.h

Ericvf, 2018-01-13:
Thanks for the feedback guys, very appreciated. It works after changing the filename. I thought I tried that yesterday but I must have made a mistake. For those interested to do with with PIO I simply used this configuration and dragged all files from this repo into the SRC directory:

```
[env:esp32dev]
platform = espressif32
framework = espidf
board = lolin32
```

Unfortunately I could not make the `make monitor` work on Windows. Running `idf_monitor.py` manually gave me a bunch of errors. But I did get to compile it, connect via the PIO Serial Monitor and pair with a Windows PC. 

My use case for this library turns out to be wrong. This makes the ESP32 act as a USB HID device where I was hoping it to act like a USB HID over Serial and pair with peripherals over bluetooth. I have some other projects where this will come in handy though to I'm very glad to have found this. Keep up the good work gentlemen.

closed 2018-01-13 by Ericvf

benjaminaigner, 2018-01-15:
> Keep up the good work gentlemen.

Thanks for the flowers :-)

If you need further examples, either use the esp-idf provided examples or ask on Github (this project is heavily based on an example from Espressif guys).

zoutepopcorn, 2018-01-22:
Great, maybe we can open a wiki, so we can document how to build, or some other documentation

So under platformio, create new esp32 project

```
|-- lib
|-- src  -- main         --   ble_hidd_demo_main.c
|        | -  makefile
|        | -  ..
|
- platformio.ini

```

If you have errors on line 29 on **ble_hid_demo_main.c**
change **eps_bt.h** 

```
#include "bt.h"
```