asterics/esp32_mouse_keyboard #6: Question about code

semigodking, 2018-03-15:
Hi Benjamin,
Would you please help me to explain the following lines of code?

For sending key code,  a zero keycode is sent after sending 'y'. Why a zero key code is needed? When is it needed?

```
                    keycode = 28;
                    esp_hidd_send_keyboard_value(hid_conn_id,0,&keycode,1);
                    keycode = 0;
                    esp_hidd_send_keyboard_value(hid_conn_id,0,&keycode,1);

```
Similar to mouse click,  a zero mouse value is sent after sending mouse click value.  Why and when is zero mouse value needed?
 ```
               case 'r':
                    esp_hidd_send_mouse_value(hid_conn_id,0x02,0,0,0);
                    esp_hidd_send_mouse_value(hid_conn_id,0x00,0,0,0);

```

Thank you!

benjaminaigner, 2018-03-15:
Hi @semigodking 

these packets are release actions. If you don't send empty reports, you will have a kind of sticky keys.
-> it seems like you press a keyboard key and a mouse button permanently.

benjaminaigner, 2018-03-15:
If you need further answers, you're welcome to ask. Otherwise please close this issue :-)

semigodking, 2018-03-15:
Thank you. One more question. Does the release mean all leading key codes
are released?

benjaminaigner, 2018-03-15:
If you look into esp_hidd_prf_api.c:109, you see that "esp_hidd_send_keyboard_value()" takes a parameter "num_key". This is the amount of keycodes which should be read. In this case, only 1 keycode is set and released.
You can pass an array of up to 6 elements which are simply mapped to a HID report.

If you want to manage it yourself, you can implement a keycode_add, keycode_remove function which changes a 6byte array and pass this array into esp_hidd_send_keyboard_value for sending updates.

Example: https://github.com/asterics/FLipMouse-esp32/blob/ecdefbed3f9aab933567ff36ca65a4319834c448/main/keyboard_layout_helper/keyboard.c#L2109

semigodking, 2018-03-15:
Thank you very much! That example really helps.

closed 2018-03-15 by semigodking