asterics/esp32_mouse_keyboard #7: ESP32 gets warm, WIFI stops working

kopchik, 2018-03-28:
Hi!

Thanks for the code, it almost works for me. There are a few issues I noticed.

1) WIFI on my laptop almost stops working when I connect to ESP32. May it happen it somehow starts transmitting all the time polluting?

2) The module becomes noticeably warm while working. I'm not sure what power consumption is, but definitely it will not last long from the battery.

3) (Not sure if relevant) I have "Call back not found for application conn_id=3" error. This happened after I disconnected and re-connected laptop. I had to do `make erase_flash`, otherwise somehow esp32 was still remembering my laptop and re-pairing/reconnecting didn't work.

## Additional info

1. ESP-IDF is latest from git
1. ESP32 module revision is 0.
1. This is the module itself

How I see wifi problem:

With esp32 (notice packet loss and max ping, this is not the worst case):
```
--- 192.168.2.1 ping statistics ---
548 packets transmitted, 483 received, 11% packet loss, time 55687ms
rtt min/avg/max/mdev = 0.732/129.743/1004.019/253.840 ms, pipe 10
```

Without (esp32 connected):
```
--- 192.168.2.254 ping statistics ---
761 packets transmitted, 760 received, 0% packet loss, time 76603ms
rtt min/avg/max/mdev = 0.820/4.813/88.536/9.019 ms
```

benjaminaigner, 2018-03-29:
Hi,

answering your questions:

1.) This ESP32 project sends only if something is received via UART (console or "make monitor").
If no input is sent, no packets are sent in this example.

2.) Please describe this issue to the esp-idf developers, I cannot do anything about low-power modes as long as it not really implemented in IDF. Maybe you try to switch on DFS in make menuconfig?

3.) I experienced this error as well, I will check on it, but currently I cannot give you advise.

Additional infos:

1.) I use the GIT version as well
2.) Same rev.
3.) Which module? I use Sparkfun's ESP32 thing.

benjaminaigner, 2018-05-08:
Closed due to no response.

closed 2018-05-08 by benjaminaigner