asterics/esp32_mouse_keyboard #8: With latest IDF 3.0, compile fails for esp_gatts_api.h and other BT headers

lunarprince, 2018-04-01:
1. IDF3.0 10.Jan.2018 toolchain is 1.22.0-80-g6c4433a, so warning is shown as required toolchain is lower: 1.22-0-8-75. Need to update to latest toolchain

WARNING: Toolchain version is not supported: 1.22.0-80-g6c4433a
Expected to see version: 1.22.0-75-gbaf03c2
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
WARNING: Toolchain version is not supported: 1.22.0-80-g6c4433a
Expected to see version: 1.22.0-75-gbaf03c2
Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.

2. esp_gatts_api.h and other bluedroid headers are not getting included as the sdkconfig has a new item CONFIG_BLUEDROID_ENABLED=y.  Need to enable this to get the files included.

CC build/main/hid_dev.o
In file included from C:/optimus/esp32/esp32_mouse_keyboard/main/hid_dev.h:18:0,
                 from C:/optimus/esp32/esp32_mouse_keyboard/main/hid_dev.c:15:
C:/optimus/esp32/esp32_mouse_keyboard/main/hidd_le_prf_int.h:19:27: fatal error: esp_gatts_api.h: No such file or directory
compilation terminated.
make[1]: *** [/c/optimus/esp-idf-v3.0-rc1/make/component_wrapper.mk:243: hid_dev.o] Error 1
make: *** [C:/optimus/esp-idf-v3.0-rc1/make/project.mk:448: component-main-build] Error 2

benjaminaigner, 2018-04-02:
Dear @lunarprince 

1.) This is not related to this project, please follow the setup instructions step by step:
https://esp-idf.readthedocs.io/en/latest/

2.) I cannot reproduce this issue, mainly because your error might be related to an OUTDATED version of the IDF. Or you didn't upgrade to the latest version of this repository.

CONFIG_BLUEDROID_ENABLED is enabled since 18days.....

https://github.com/asterics/esp32_mouse_keyboard/blob/ae191af2b42deaa308a164929fc0b9145add4319/sdkconfig#L129

benjaminaigner, 2018-04-05:
I will close this issue, please re-open if necessary.

closed 2018-04-05 by benjaminaigner