asterics/esp32_mouse_keyboard #9: Windows 10 does not register key reports

Galzai, 2018-04-28:
It seems that while the ESP32 manages to properly connect to windows, key reports do not seem to register. the log does not seem any different from the log while sending reports to Linux/Android.  I think this might be related to the device descriptor windows requires:
from QMK :
"Mac OS-X and Linux automatically load the correct drivers. On Windows, even though  driver is supplied by Microsoft, an INF file is needed to load the driver. " 

https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/pjrc/usb.c

benjaminaigner, 2018-05-02:
[@Galzai 

In general you are right, AFAIK an INF file is not necessary if you connect a Bluetooth mouse.
Your link is also targeting USB connected devices.
If you have a look at our [FLipMouse](https://github.com/asterics/FLipMouse)/[FABI](https://github.com/asterics/FABI) devices, we never needed an INF file for basic mouse/keyboard/joystick operation. Only for the serial device.

chegewara, 2018-05-04:
Android (im assuming linux too) can connect to HOG device if input report pipe number is not set. According to hid documentation in this case its used default value 0, but it wont work with windows. Windows requires that any value needs to be set. It took me couple weeks to figure it out.

Its this line:
https://github.com/asterics/esp32_mouse_keyboard/blob/newBLE/components/nkolban_BLE/HID_kbdmousejoystick.cpp#L371

chegewara, 2018-05-07:
Well, its time to "get hands dirty", compile app and test it in real life.

chegewara, 2018-05-07:
First try and is connected on windows 10, but after some time is driver error, which mean there is issue with report map.

chegewara, 2018-05-07:
> Although keyboard is working, mouse is not registered on Win10. It is working on Linux & Android. Do you have any suggestions where to have a look on Win10? There is no driver error or anything similar, it just doesn't work..

I have no idea what wrong, but mouse is working on my asus windows 10, cursor is randomly jumping on my screen.

BTW i have changed to 
		`pSecurity->setCapability(ESP_IO_CAP_NONE);`
and it works

chegewara, 2018-05-07:
@benjaminaigner 
This is parsed report map. I see you have 8 buttons, most likely its not an issue(it works on my windows 10), but could you try with standard 3 buttons mouse?
```
0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x06,        // Usage (Keyboard)
0xA1, 0x01,        // Collection (Application)
0x85, 0x01,        //   Report ID (1)
0x75, 0x01,        //   Report Size (1)
0x95, 0x08,        //   Report Count (8)
0x05, 0x07,        //   Usage Page (Kbrd/Keypad)
0x19, 0xE0,        //   Usage Minimum (0xE0)
0x29, 0xE7,        //   Usage Maximum (0xE7)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0x01,        //   Logical Maximum (1)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x01,        //   Report Count (1)
0x75, 0x08,        //   Report Size (8)
0x81, 0x03,        //   Input (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x05,        //   Report Count (5)
0x75, 0x01,        //   Report Size (1)
0x05, 0x08,        //   Usage Page (LEDs)
0x19, 0x01,        //   Usage Minimum (Num Lock)
0x29, 0x05,        //   Usage Maximum (Kana)
0x91, 0x02,        //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x95, 0x01,        //   Report Count (1)
0x75, 0x03,        //   Report Size (3)
0x91, 0x03,        //   Output (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x95, 0x06,        //   Report Count (6)
0x75, 0x08,        //   Report Size (8)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0x68,        //   Logical Maximum (104)
0x05, 0x07,        //   Usage Page (Kbrd/Keypad)
0x19, 0x00,        //   Usage Minimum (0x00)
0x29, 0x68,        //   Usage Maximum (0x68)
0x81, 0x00,        //   Input (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection
0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x02,        // Usage (Mouse)
0xA1, 0x01,        // Collection (Application)
0x85, 0x02,        //   Report ID (2)
0x09, 0x01,        //   Usage (Pointer)
0xA1, 0x00,        //   Collection (Physical)
0x05, 0x09,        //     Usage Page (Button)
0x19, 0x01,        //     Usage Minimum (0x01)
0x29, 0x08,        //     Usage Maximum (0x08)
0x15, 0x00,        //     Logical Minimum (0)
0x25, 0x01,        //     Logical Maximum (1)
0x95, 0x08,        //     Report Count (8)
0x75, 0x01,        //     Report Size (1)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x01,        //     Usage Page (Generic Desktop Ctrls)
0x09, 0x30,        //     Usage (X)
0x09, 0x31,        //     Usage (Y)
0x09, 0x38,        //     Usage (Wheel)
0x15, 0x81,        //     Logical Minimum (-127)
0x25, 0x7F,        //     Logical Maximum (127)
0x75, 0x08,        //     Report Size (8)
0x95, 0x03,        //     Report Count (3)
0x81, 0x06,        //     Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              //   End Collection
0xC0,              // End Collection
```

benjaminaigner, 2018-05-08:
Hi, thank you very much for testing.

It tried it again right now with another notebook of my colleague, it works with mouse / keyboard in my current configuration (mouse with 8 buttons, ESP_IO_CAP != NONE).

I will change the report to use 3 buttons and add the 5bit padding.

benjaminaigner, 2018-05-14:
@chegewara 

I've figured out where the difference on Win10 is:

- **Win 10 Education N**: does NOT work with ESP_IO_CAP_NONE
- **Win 10 Education N**: does work with ESP_IO_CAP_NONE
- As you reported, Pro versions work anyway.

Maybe the **N** stands for NOT working :-)? It is clearly related to the IO capabilities, with NONE it does not even connect properly...

Here is a log file for a non-working connection: https://pastebin.com/eWHYfp4y
Same notebook & example works, if I use ESP_IO_CAP_OUT.

chegewara, 2018-05-14:
Sadly i have only one laptop and i cant perform tests on any other device. I've read that windows requires bonded connection thats why examples are prepared to connect with pairing, even if initialy it is not used.
Im glad you could confirm that windows version  makes difference, this is another step in resolving ble hid issues.

benjaminaigner, 2018-05-15:
I can confirm that this is a Windows issue, which can be resolved by using the old-style control center for pairing:

https://superuser.com/a/1245995

I don't know why this only happens on Education N, but at least there is a possibility to bypass this issue.

closed 2018-09-22 by Galzai