asterics/esp32_mouse_keyboard #11: Reconnection takes very long and seems unstable

ChrisVeigl, 2018-06-01:
I was testing the Bluetooth connection (in newBLE branch with BT-stack by neil kolban). 
It seems that the reconnection takes very long  (up to 10 minutes !) if the ESP32 was connected to a BT-client but got out-of-range or was restarted. (The devices were paired and connected before - then the clients was moved out-of-range and brought back several times.)

Shortly after the devices get in range after disconnection, the "client connected" message appears - but the connection breaks down immediatley with "reason=0x003d" and advertising is restarted. Then it takes ages .... and then the client reconnects successfully.  see screenshot:

[image]

This is repeatable and happens every time when the connection is lost.

Sometimes another problem occurred: after a connection is lost and is re-established, the client (in this case tested with an Android Nexus 5X phone) claims to be "connected" but the registration of the BT-services never completes successfully (so no mouse/keyboard interaction is possible).  when this happens, also a reset of the esp32 + reboot of the client does not help -  can be cured only by unpairing the devices and starting a new pairing sequence (!)

ChrisVeigl, 2018-07-01:
I performed some more tests of the pairing / reconnection problems on different platforms
(two different Lenovo laptops, two different Android phones and an iPhone)
The results are different:

** Lenovo Yoga Pro Utrabook / Win 8.1 / Intel Wireless Bluetooth 4.0 + HS-Adapter: **
pairing does not work at all (message:"that didnt work ..." - maybe incompatibility with Win8 because win8 needs pin code? - see https://superuser.com/questions/847605/why-wont-some-bluetooth-keyboards-pair-in-windows-8-1)

** Lenovo Thinkpad P51 / Win 10 professional / Intel Wireless Bluetooth + BLE enumaeration & GATT: **
pairing works 
reconnection takes very long (from 20 seconds to ~8 minutes, usually about 2-3  minutes) 
2 times during operation, the BT device on the laptop disapperared suddenly (driver crash??), computer needed to be restarted to get a working BT-device again (!) 

** Samsung galaxy A3 / Android 7.0 / Kernel 3.10.61: **
pairing works
reconnection takes very long (from 20 seconds to ~8 minutes, usually about 2-3  minutes) 

** LG Galaxy Nexus-5s / Android 8.1 / Kernel 3.10.73: **
pairing works
reconnection works if esp-side drops the connection (e.g. esp reset) 
reconnect takes very long (from 20 seconds to ~8 minutes, usually about 2-3  minutes) 
reconnection does not work if connection is dropped from other end (e.g. phone switched off / restarted)

** iPhone 6S / iOS 11.3 **
pairing works
reconnection works if esp-side drops the connection (taking only several seconds, up to 30 seconds)
reconnection does not work if connection is dropped from other end (e.g. phone switched off / restarted / BT-turned off+on)
(in this case the state "connected" is shown, but no HID activity is working - only way to get it working again is to remove the bt-device and renew the pairing)

chegewara, 2018-07-09:
Since its about this about this issue i will not answer on Kolban's repo. I have esp-idf master and i have pulled fresh copy from this repository. I did change only 3 things:
- commented out this line to get more logs just in case: `    // esp_log_level_set("*", ESP_LOG_INFO); 
`
- in menuconfig i did turn off Classic Buetooth (because i dont understand the reason, why its turned on),
- for debugging reasons i did change all log levels for bluetooth in menuconfig,

My observations:
- reconnecting is taking few seconds (because high debug level),
- no issues with reconnecting during testing with: windows 10 laptop, samsung s9+, i have problem to connect with samsung s5 but im assuming its normal behavior since security has been added to make code compatible with iPhones and samsung s5 has ble 4.0 which is too old to work with secured devices.

Here you can see my logs when i uncommented that line `esp_log_level_set("*", ESP_LOG_INFO); `:
```
I (33427) HAL_BLE: Client disconnected, restarting advertising
I (33427) BT_APPL: BTA_DmSetBleAdvParamsAll: 40, 80

I (33427) BT_APPL: adv_type = 0, addr_type_own = 0, chnl_map = 7, adv_fil_pol = 0

I (33437) BT_APPL: BTA_DmSetBleAdvParamsAll: 40, 80

I (33437) BT_APPL: adv_type = 0, addr_type_own = 0, chnl_map = 7, adv_fil_pol = 0

E (33457) BT_APPL: bta_dm_ble_set_adv_params_all(), fail to start ble adv.
I (33457) BT_L2CAP: L2CA_SetDesireRole() new:x1, disallow_switch:0
I (47987) BT_GATT: GATT_GetConnIdIfConnected status=1   // <--- start reconnecting

I (47997) HAL_BLE: Client connected !
I (47997) BT_L2CAP: L2CA_SetDesireRole() new:x1, disallow_switch:0
I (48577) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (48577) BT_GATT: GATTS_SendRsp: conn_id: 3  trans_id: 1  Status: 0x0000

I (48577) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 7491ae6fee99
I (48677) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (48677) BT_GATT: GATTS_SendRsp: conn_id: 3  trans_id: 2  Status: 0x0000

I (48677) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 7491ae6fee99
I (48767) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (48777) BT_GATT: GATTS_SendRsp: conn_id: 3  trans_id: 3  Status: 0x0000

I (48777) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 7491ae6fee99
I (48867) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (48867) BT_GATT: GATTS_SendRsp: conn_id: 3  trans_id: 4  Status: 0x0000

I (48877) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 7491ae6fee99
I (48967) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (48967) BT_GATT: GATTS_SendRsp: conn_id: 3  trans_id: 5  Status: 0x0000

I (48967) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 7491ae6fee99
I (49067) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (49067) BT_GATT: GATTS_SendRsp: conn_id: 3  trans_id: 6  Status: 0x0000

I (49067) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 7491ae6fee99
I (49157) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (49167) BT_GATT: GATTS_SendRsp: conn_id: 3  trans_id: 7  Status: 0x0000

I (49167) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 7491ae6fee99
I (49257) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (49257) BT_GATT: GATTS_SendRsp: conn_id: 3  trans_id: 8  Status: 0x0000

I (49257) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 7491ae6fee99
I (49357) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (49357) BT_GATT: GATTS_SendRsp: conn_id: 3  trans_id: 9  Status: 0x0000

I (49357) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 7491ae6fee99
I (49457) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (49457) BT_GATT: GATTS_SendRsp: conn_id: 3  trans_id: 10  Status: 0x0000

I (49457) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 7491ae6fee99
I (49547) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (49557) BT_GATT: GATTS_SendRsp: conn_id: 3  trans_id: 11  Status: 0x0000

I (49557) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 7491ae6fee99
I (49647) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (49647) BT_GATT: GATTS_SendRsp: conn_id: 3  trans_id: 12  Status: 0x0000

I (49647) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 7491ae6fee99
I (49747) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (49747) BT_GATT: GATTS_SendRsp: conn_id: 3  trans_id: 13  Status: 0x0000

I (49747) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 7491ae6fee99
```

When we have less logs in console connecting/reconnecting is very fast (about 2-3 sec). To reconnect i did turn off/on bluetooth in my smartphone and laptop.
Board i am doing tests is devKitC v2 (its old board). So, my only suggestion is to turn off classic BT, and maybe (just in case) to change bluetooth stack to 6000-8000 (but only just in case).

ChrisVeigl, 2018-07-09:
dear Che,

thank you for your investigation and help!
i made a fresh pull of the esp-idf master and changed the sdkconfig so that CONFIG_CLASSIC_BT_ENABLED is switched off. My first tests looked promising as my Andriod Nexus 5 could re-establish the connection after BT was turned off and on again. It also worked after reboot.
The reconnection still took very long (about 3-4 minutes in these trials) but it worked.

Then it set CONFIG_CLASSIC_BT_ENABLED=y again (to verify that this is the reason and not some change which i pulled from the esp-idf) and indeed: reconnection did not work again in case BT was turned off/on or the phone was restarted.

Now the strange thing: i turned CONFIG_CLASSIC_BT_ENABLED off again, rebuilt, but still i cannot get the reconnection working! i waited 20 minutes, tried it several times in different szenarios, but no success.... same problem as before - no reconnect when client drops connection. really weird. 

The pairing problem with win8.1 persists (probably the reason is the pin code entry)

I could not try any iOS device by now.

ChrisVeigl, 2018-07-15:
some news: i continued tests with Android (Nexus 5) and iOS (iPhone 6s) today.
although no difference in the settings, the Android client could connect multiple times after BT was switched off/on or the phone was restarted. sometimes it took very long (5-6 minutes) sometimes only 30 seconds.  i have no idea what causes these huge differences or why it did not work at all when it tested the last time (i waited 20 minutes...).

Regarding the iPhone - the problem still persists: first time a client is paired / connected, Keyboard reports are processed correctly, even when the server (ESP32) is reset the reconnection (and keyboard events) work after several seconds. But when the phone drops the connection (BT switched off/on or phone restarted) the HID actions do not work after reconnection (ESP thinks it's connected but no keys are processed on the iPhone).

Something worth noting: in the logs we get multiple GATT_INSUF_ENCRYPTION errors when such a failing reconnection takes place. (This error message is just logged once when a connection is accomplished successfully). 

see attached logs:
[Log_esp_reconnection-iOS.txt](https://github.com/asterics/esp32_mouse_keyboard/files/2196011/Log_esp_reconnection-iOS.txt)

chegewara, 2018-07-15:
it can be related, but im not sure:
https://github.com/espressif/esp-idf/issues/2036

BillyNate, 2018-11-08:
I uncommented the `setInitEncryptionKey` line in [`HID_kbdmousejoystick.cpp`](https://github.com/asterics/esp32_mouse_keyboard/blob/master/components/nkolban_BLE/HID_kbdmousejoystick.cpp#L665), this seemed to solve the issue.

benjaminaigner, 2020-02-13:
Fixed with #27

closed 2020-02-13 by benjaminaigner

DXK87, 2020-05-22:
I have the same problem with the ESP32, it made the connection with the classic bluetooth, and when I get too far away and lose the signal it disconnects and blocks the ESP32, it can only reconnect when the ESP32 restarts. any solution to the issue?

benjaminaigner, 2020-05-22:
Dear @DXK87 
just to be sure: we do not used BT Classic here, this is a Bluetooth Low Energy implementation
(HID over GATT -> HoG).
Do you have this problem with our project?