asterics/esp32_mouse_keyboard #15: What's the best bluetooth module for a DIY keyboard? BC417(HC-05), CC2540, CC2650, or ESP32?

coreboot2, 2018-11-27:
What's the best bluetooth module for a DIY keyboard? BC417(HC-05), CC2540, CC2650, or ESP32?

I'm thinking about making a DIY bluetooth keyboard.

I searched and found many different modules that can do this. For example:

One is the BC417 (HC-05) module: https://mitxela.com/projects/bluetooth_hid_gamepad

Another one is the CC2540 module: https://hackaday.com/2017/05/20/the-tiniest-mechanical-keyboard-ever/
Or the newer CC2650 :https://e2e.ti.com/support/wireless-connectivity/bluetooth/f/538/t/659154

Are there anyother modules that can be used for a keyboard not listed? And how to choose, which one is the best? What are the advantages or disadvantages?

I also want the keyboard to have a function like the logitech's easy switch https://www.logitech.com/en-us/product/illuminated-keyboard-for-mac-ipad-iphone (using the same keyboard to pair with different computers and switch among them with one click). So can these modules do this?

13080433, 2019-02-13:
Could you tell me on which basis you had selected a device from all of the listed above?
Or what are the limitations for Bluetooth HID?

pavrda, 2019-10-10:
I vote for NRF51 or NRF51 . Very open and friendly platform for developers.

benjaminaigner, 2020-02-13:
I think everything based on CCxxxx is not really useful. They are very cheap, but you need a crappy & very expensive compiler (sdcc would be possible, but not the best solution). Before starting this project, I tried to re-flash a HM10 module with BLE-HOG, but this was not really successful.
Also, the community for TI products is.... Let's say not very busy&helpful, especially for older chips.

I've never used the BC417.

The ESP32 has some drawbacks, but all in all it is a useful product (except there is no version with BLE AND USB :-)).

NRF52 is also a good choice, although the "SoftDevice" is an annoying piece of proprietary blob...

closed 2020-02-13 by benjaminaigner