asterics/esp32_mouse_keyboard #16: nkolban_BLE with some errors

HotIce0, 2018-12-03:
```
HotIce0-mac:esp32_mouse_keyboard HotIce0$ make flash
Toolchain path: /Users/HotIce0/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a
Compiler version: 5.2.0
Python requirements from /Users/HotIce0/esp/esp-idf/requirements.txt are satisfied.
CXX build/nkolban_BLE/BLEUtils.o
/Users/HotIce0/esp/esp32_mouse_keyboard/components/nkolban_BLE/BLEUtils.cpp: In static member function 'static std::__cxx11::string BLEUtils::gattServerEventTypeToString(esp_gatts_cb_event_t)':
/Users/HotIce0/esp/esp32_mouse_keyboard/components/nkolban_BLE/BLEUtils.cpp:988:8: error: enumeration value 'ESP_GATTS_SEND_SERVICE_CHANGE_EVT' not handled in switch [-Werror=switch]
  switch(eventType) {
        ^
cc1plus: some warnings being treated as errors
make[1]: *** [BLEUtils.o] Error 1
make: *** [component-nkolban_BLE-build] Error 2
```

So, I add the code to handled.

```C
/**
 * @brief Return a string representation of a GATT server event code.
 * @param [in] eventType A GATT server event code.
 * @return A string representation of the GATT server event code.
 */
std::string BLEUtils::gattServerEventTypeToString(esp_gatts_cb_event_t eventType) {
	switch(eventType) {
	case ESP_GATTS_REG_EVT:
		return "ESP_GATTS_REG_EVT";
		
	case ESP_GATTS_SEND_SERVICE_CHANGE_EVT:
		return "ESP_GATTS_SEND_SERVICE_CHANGE_EVT";
```

coreboot2, 2018-12-15:
I get this error, is it the same one as this? How to fix it?

```
C:/msys32/home/coreboot2/esp/esp32_mouse_keyboard/components/nkolban_BLE/BLEUtils.cpp:                                                                  In static member function 'static std::__cxx11::string BLEUtils::gattServerEvent                                                                 TypeToString(esp_gatts_cb_event_t)':
C:/msys32/home/coreboot2/esp/esp32_mouse_keyboard/components/nkolban_BLE/BLEUtils.cpp:992:7: error: 'ESP_GATTS_SEND_SERVICE_CHANGE_EVT' was not declared in this scope
  case ESP_GATTS_SEND_SERVICE_CHANGE_EVT:
       ^
make[1]: *** [/home/coreboot2/esp/esp-idf/make/component_wrapper.mk:286：BLEUtils.o] Error 1
make: *** [C:/msys32/home/coreboot2/esp/esp-idf/make/project.mk:468：component-nkolban_BLE-build] Error 2
```

benjaminaigner, 2018-12-18:
@coreboot2 

thanks for reporting this issue.
Please try to update the esp-idf, it seems this is a new event, which was introduced with a new version (would match with @HotIce0 error of an unhandled event type).

thedevleon, 2019-01-13:
> I get this error, is it the same one as this? How to fix it?
>
> ``` [...]

You need to comment out the two lines referencing ESP_GATTS_SEND_SERVICE_CHANGE_EVT in components/nkolban_BLE/BLEUtils.cpp

benjaminaigner, 2020-02-13:
Irrelevant with #27

closed 2020-02-13 by benjaminaigner