asterics/esp32_mouse_keyboard #19: Test y keycode does not registers on android

sandric, 2019-02-24:
I played with demo code a bit, after compilation - I can see `FLIPMOUSE` device appeared my android phone, but entering `q` to send `y` keycode seems to make nothing. Although I see it sends it in serial monitor: before connection appears I see:

```
Serial_client > q
I (15574) CONSOLE_UART: Not connected, ignoring 'q'

Serial_client > I (15574) CONSOLE_UART: Not connected, ignoring '

'
```
after esp32 paired with my phone I see following:
```
Serial_client > q
I (124554) CONSOLE_UART: received q: sending key y for test purposes

Serial_client > I (124554) HAL_BLE: Keyboard received: 2/28

I (124554) CONSOLE_UART: received: 10
```
but no `y` appears as input, I tried in notes and sms apps - the same. Does output looks right, especially  "2/28" part?

Also no matter what I tried - my windows 10 laptop does not see esp32 as bt device to connect, it sees android phone where esp32 is visible with no problem though.

I didn't changed demo code, all I did is changed `python` to `python2` in sdkconfig file, I'm on archlinux virtual machine, and it compiles with no problems. Running `make flash` first time though it asks me ~20 questions about settings, but all I did is pressed enter to all of them for it to take default value.

sandric, 2019-02-24:
Forgot to add, I'm using `LuaNode32` based on `ESP-WROOM-32` kit, here's the [image](https://arduino.ua/products_pictures/large_AOC515-1.jpg)

benjaminaigner, 2019-02-25:
Dear @sandric 

Console output looks quite normal.
"2/28" means: type of action is 2 (enum) -> PRESS_RELEASE (sending a press report followed by a release report). 28 is the keycode, according to keyboard.h it is "Y".

Do you use a different keyboard layout on your android phone?

Is it possible to post a full debug log?

benjaminaigner, 2020-02-13:
@sandric 
Just tested it with the current master (esp-idf: v4.1) and I receive the keyboard key on an Android (LineageOS 16.0, Android 9).
If the problem persists, please reopen this issue.

closed 2020-02-13 by benjaminaigner