asterics/esp32_mouse_keyboard #20: Sending mouse commands

bilogic, 2019-04-20:
Hi,

In the ```readme.md```, it was stated that Mouse command starts with a command byte that needs to be set to 'M', however I don't see that in the definition. Can explain and point out to me where 'M' is inserted? Thank you.

```
typedef struct mouse_command
{
    int8_t x;
    int8_t y;
    int8_t wheel;
    uint8_t buttons;
} mouse_command_t;
```

ChrisVeigl, 2019-04-20:
Hi,

the commands are sent to the ESP32 via the second UART (UART2, TX = GPIO17 and RX = GPIO16). 
You can do this via another microcontroller or via an USB-to-serial converter (3,3V). 
The 'M' command is processed around

https://github.com/asterics/esp32_mouse_keyboard/blob/bfeeec35a942258bb2b185b74689acb7802e365a/main/ble_hidd_demo_main.cpp#L328

bilogic, 2019-04-20:
Hi,

Thanks, I get it now. I was actually after how to programmatically send mouse commands. Now I know that this 'M' command/UART2 is another source of input for the mouse/keyboard.

Also, not sure if ```buttons``` was intentionally ordered to be the last, but I was very confused why it worked,

https://github.com/asterics/esp32_mouse_keyboard/blob/625440c04d0536783f51585b9161a9ae42c24686/components/nkolban_BLE/HID_kbdmousejoystick.h#L99-L104

as ```esp_ble_gatts_send_indicate()``` accepts only a struct as follows:
```
typedef struct mouse_command
{
	uint8_t buttons;	// mouse_state[0]
	int8_t x;			// mouse_state[1]
	int8_t y;			// mouse_state[2]
	int8_t wheel;		// mouse_state[3]
} mouse_command_t;
```

Now I know the re-ordering took place here:
https://github.com/asterics/esp32_mouse_keyboard/blob/625440c04d0536783f51585b9161a9ae42c24686/components/nkolban_BLE/HID_kbdmousejoystick.cpp#L340-L358

Thanks!

closed 2019-04-20 by bilogic