asterics/esp32_mouse_keyboard #23: esp-idf:esp_ble_gap_update_whitelist has an update

rm3n, 2019-05-26:
appears not to be working with latest esp-idf; seems declaration for esp_ble_gap_update_whitelist used in [BLEDevice.cpp] got updated late march, tried supplying BLE_WL_ADDR_TYPE_PUBLIC, how ever I was not able to make this work.

benjaminaigner, 2019-05-27:
@rm3n 
Thanks for the information!
It's a little bit annoying that the IDF's APIs are changed all the time :-).

Does it work for a release version? We always used v3.xxx

Greetings

rm3n, 2019-05-29:
@benjaminaigner yea, I come back every 6 months to work on this processor and nothing works lol; so I guess you meant to try an older version; I wasn't sure which version; 

I managed to get it to work on idf|release/v3.0; had to remove ESP_GATTS_SEND_SERVICE_CHANGE_EVT from nkolban|BLEUtils.cpp to pass the build warn error and got it to work :+1: 

the new release/v3.3 has so many fixes; how ever I think wifi stack is acting little strange... though I was not able to get this work, always ends up gdb.

I was trying to see if there was an update to nkolban's ble on the original branch or arduino version; no success there, also component may need to be imported as submodule?

copercini, 2019-06-06:
@rm3n here it works with `BLE_WL_ADDR_TYPE_RANDOM` instead `BLE_WL_ADDR_TYPE_PUBLIC`

benjaminaigner, 2020-02-13:
Dear @rm3n @copercini ,
I've changed the HOG implementation to a new version, based on the Espressif example.
I think this error should not happen there. If yes, please re-open this issue.

> @benjaminaigner yea, I come back every 6 months to work on this processor and nothing works lol; so I guess you meant to try an older version; I wasn't sure which version;

Same here, REALLY annoying...  I have at least 4 versions of esp-idf + xtensa-gcc installed :-)

closed 2020-02-13 by benjaminaigner