asterics/esp32_mouse_keyboard #24: Not enough poll rate with mouse commands

n1rml, 2019-11-18:
I have made an air mouse using your code and MPU6050 (6 axis gyro and accelerometer)
the sensor is able to report at 100hz, but when i send mouse commands at higher speeds than 10hz the commands get queued and i get heavy latency (near 1 sec), but at slow speeds (10hz or lower) there is fairly low latency but the cursor movements are very choppy.
So, i wanted to know if it's **possible to increase the report rate of mouse commands in your code ?**
i'm a total newbie to all this and need help [here's](https://github.com/n1rml/esp32_airmouse) my repo.

and thank you for making this, its awesome

benjaminaigner, 2019-11-18:
Dear @n1rml 
congrats on your first ESP-IDF project :+1: 

A few suggestions:
-) You should combine x/y values into one queue element, if both of them are != 0.
https://github.com/n1rml/esp32_airmouse/blob/a1251c1bfad4deed316e85b40dfe7fd1fd7f0d48/main/main.cpp#L289
This way, it's possible to cut down the sent packages to the half.
-) Personally, I don't think 100Hz are possible. I think USB HID allows up to 1kHz, but I don't think this is possible with BLE (I don't have a source for this, but I think the latency for each packet is a few ms and you need quite a few for one HID packet).
-) The "choppy" mouse speed is related to the `MOUSE_SPEED`define. You move the mouse cursor either 0,30,60 or 90 pixels each time. Try to reduce this value and use more range from the IMU.

n1rml, 2019-11-19:
Thanks a lot for those tips, i will apply them and update you.
i have kept the  MOUSE_SPEED at 30 because at lower values it's tough to control the cursor and the cursor movements are between 0.30 - 60 pixels , so i'm not sure how changing this range will help.

n1rml, 2019-11-19:
[image]
I combined the x/y values and increased the report rate to 20hz, it's a lot better now.
thanks
in this screen capture the cursor is moving using the esp32.

n1rml, 2019-11-23:
Hello @benjaminaigner can you tell me about BLE HID implementation in FlipMouse-esp32 (i.e. report rate and effective latency)

benjaminaigner, 2020-02-13:
Dear @n1rml 
unfortunately, I got no e-mail notification on your question.

We use basically the same firmware in the FLipMouse and here.

The FlipMouse calculates the mouse movement by:

- Reading the ADC values & subtracting the zeropoint (determined by calibration command or on startup)
- Multiplying x/y axis with the sensitivity value
- Increase the output each measurement (acceleration) & limit to upper value
- Send x/y values to Bluetooth/USB HID

If the calculated movement is higher than 127 (limit of signed 8bit), we split it into more packets.

We experienced similar speed related "hiccups", but I'm not sure where the problem is located.
Although Neil Kolbans stack was a wonderful start for this project, I switched back to pure C based code (taken from the Espressif HOG example).

If you want to give it another try, use current master branch. Maybe it works better for you.

Your project is interesting, maybe I ask students of mine to build an airmouse here :-).

If you need further infos, just write here. Otherwise it would be nice to close this issue.

Greetings,
Benjamin

benjaminaigner, 2020-02-13:
@n1rml THX for closing :-)

closed 2020-02-13 by benjaminaigner