asterics/esp32_mouse_keyboard #26: How to turn on joystick mode and change keyboard keys? I cant find

akulij, 2020-02-08:


akulij, 2020-02-08:
i found where to change the keys for the keyboard and mouse? but I can't switch the mode to the gamepad

benjaminaigner, 2020-02-10:
Dear @akulij ,
Joystick is currently not implemented in Bluetooth.
If you have the possibilities to support us with this task, it would be greatly appreciated.

The Joystick report is missing in the HID descriptors and the BLE services.

Greetings

akulij, 2020-02-10:
I found the gamepad and joystick apperenace, and it seems to work, but I don’t know what data to send, because I don’t have a gamepad to intercept
[image]
[image]

benjaminaigner, 2020-02-11:
Do you use the master or newBLE branch?
In the newBLE branch, I switched back to plain C code without Neil Kolbans library.

Anyway, in the master branch, you need to implement this task:
https://github.com/asterics/esp32_mouse_keyboard/blob/bfeeec35a942258bb2b185b74689acb7802e365a/components/nkolban_BLE/HID_kbdmousejoystick.cpp#L364

It is necessary to fill up the HID report with the corresponding bytes (as defined in the HID descriptor here:
https://github.com/asterics/esp32_mouse_keyboard/blob/bfeeec35a942258bb2b185b74689acb7802e365a/components/nkolban_BLE/HID_kbdmousejoystick.cpp#L173
E.g.: first 4 bytes (32bits) are the joystick buttons.

benjaminaigner, 2020-02-13:
Dear @akulij 
as we didn't use the master branch for a longer time, we merged the newBLE branch now.
There is also no joystick support right now.

If you want joystick, you could try with the latest version of the master branch with the Neil Kolban stack (I created a pre-release to find this easily) or you could try it with the current master.

In my opinion it would be better to do this on the current master, as we need the joystick functionality there as well. With newer devices, we also experienced major connection problems with the previous stack.

benjaminaigner, 2023-02-16:
Dear @akulij ,

on the current master branch, gamepad is working now (tested with Win10, Linux & Android).

Note: iOS doesn't work with configured gamepad, I will add a possibility to enable the gamepad on demand when it is added (idf.py menuconfig).

Note2: please use esp-idf v5.0, v4.x does not work anymore & esp-idf master has currently a broken BLE implementation.

closed 2023-02-16 by benjaminaigner