asterics/esp32_mouse_keyboard #29: Win 10/iOS

vignesh-mpc, 2020-02-13:
Reading through the closed issues and I saw that iOS and Win 10 support has been definitely working at some point, but I wasn't able to get it to work with the latest commit on master. Going to try on Android next and will report back, but wanted to see if anyone had a similar experience or how I can go about debugging this.

Thanks!

vignesh-mpc, 2020-02-13:
The test character 'q' works and correctly sends 'y' to my Android device. The mouse arrow keys also work. However, within about a minute of the connection, the ESP32 prints out the ESP_HIDD_EVENT_BLE_DISCONNECT for some reason.

Anyone else experiencing this? Happening consistently on commit f7ab88ae7a9ba4b450296a1b7970a30fa985b622

benjaminaigner, 2020-02-17:
I cannot confirm this issue for my Android device, no problem with re-connections:
```
I (442) heap_init: Initializing. RAM available for dynamic allocation:
I (448) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (454) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (460) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (467) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (473) heap_init: At 3FFC9088 len 00016F78 (91 KiB): DRAM
I (479) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (485) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (492) heap_init: At 40093A10 len 0000C5F0 (49 KiB): IRAM
I (498) cpu_start: Pro cpu start user code
I (516) spi_flash: detected chip: generic
I (517) spi_flash: flash io: dio
I (517) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (587) MAIN: loading configuration from NVS
I (587) MAIN: bt device name is: FLIPMOUSE
E (587) MAIN: error reading NVS - locale, setting to US_INTERNATIONAL
I (597) BTDM_INIT: BT controller compile version [5fda27f]
I (607) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (717) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 1, 0
I (1107) HIDD: MAIN finished...
I (1107) UART: console UART processing task started
I (1107) EXT_UART: external UART processing task started
I (5517) hal_ble: ESP_HIDD_EVENT_BLE_CONNECT
I (5807) hal_ble: remote BD_ADDR: xxxxd4eexxxx
I (5807) hal_ble: address type = 1
I (5807) hal_ble: pair status = success
I (8017) CONSOLE_UART: mouse: a
I (8717) CONSOLE_UART: mouse: d
I (8997) CONSOLE_UART: mouse: w
I (44697) CONSOLE_UART: mouse: a
I (513497) CONSOLE_UART: mouse: a
I (515837) CONSOLE_UART: mouse: d
I (524337) CONSOLE_UART: mouse: w
I (526587) CONSOLE_UART: mouse: d
I (539827) CONSOLE_UART: mouse: w
I (596277) CONSOLE_UART: mouse: s
```
esp-idf v4.1, Android 9 / LineageOS 16.0

Also no problem with Linux (Debian Stable).
I tried it with an iPad Air 2 (iOS 13.3.1) as well, keyboard input works.
For the mouse cursor, you have to activate "Assistive Touch", then the mouse cursor works immediately.

@vignesh-mpc For further assistance, please provide logs and system information.

benjaminaigner, 2020-06-29:
Closed with #35

closed 2020-06-29 by benjaminaigner