asterics/esp32_mouse_keyboard #31: Cannot build using cmake

n1rml, 2020-03-04:
Hello @benjaminaigner, i was trying to build the latest master branch code (new BLE stack) using cmake but i'm not sure about what changes are need to make it compatible with cmake or is it even possible ?

Thanks for the reply on last issue
sorry for such silly question

n1rml, 2020-03-05:
I am not able to identify what the problem is, as the ble_hid folder is included in the components section
`Component paths: C:/Users/G/esp/esp-idf/components/app_trace C:/Users/G/esp/esp-idf/components/app_update C:/Users/G/esp/esp-idf/components/asio C:/Users/G/esp/esp32_mouse_keyboard/main/ble_hid C:/Users/G/esp/esp-idf/components/bootloader C:/Users/G/esp/esp-idf/components/bootloader_support C:/Users/G/esp/esp-idf/components/bt C:/Users/G/esp/esp-idf/components/coap C:/Users/G/esp/esp-idf/components/console C:/Users/G/esp/esp-idf/components/cxx C:/Users/G/esp/esp-idf/components/driver C:/Users/G/esp/esp-idf/components/efuse C:/Users/G/esp/esp-idf/components/esp-tls C:/Users/G/esp/esp-idf/components/esp32 C:/Users/G/esp/esp-idf/components/esp_adc_cal C:/Users/G/esp/esp-idf/components/esp_common C:/Users/G/esp/esp-idf/components/esp_eth C:/Users/G/esp/esp-idf/components/esp_event C:/Users/G/esp/esp-idf/components/esp_gdbstub C:/Users/G/esp/esp-idf/components/esp_http_client C:/Users/G/esp/esp-idf/components/esp_http_server C:/Users/G/esp/esp-idf/components/esp_https_ota C:/Users/G/esp/esp-idf/components/esp_https_server C:/Users/G/esp/esp-idf/components/esp_local_ctrl C:/Users/G/esp/esp-idf/components/esp_ringbuf C:/Users/G/esp/esp-idf/components/esp_rom C:/Users/G/esp/esp-idf/components/esp_websocket_client C:/Users/G/esp/esp-idf/components/esp_wifi C:/Users/G/esp/esp-idf/components/espcoredump C:/Users/G/esp/esp-idf/components/esptool_py C:/Users/G/esp/esp-idf/components/expat C:/Users/G/esp/esp-idf/components/fatfs C:/Users/G/esp/esp-idf/components/freemodbus C:/Users/G/esp/esp-idf/components/freertos C:/Users/G/esp/esp-idf/components/heap C:/Users/G/esp/esp-idf/components/idf_test C:/Users/G/esp/esp-idf/components/jsmn C:/Users/G/esp/esp-idf/components/json C:/Users/G/esp/esp-idf/components/libsodium C:/Users/G/esp/esp-idf/components/log C:/Users/G/esp/esp-idf/components/lwip C:/Users/G/esp/esp-idf/components/mbedtls C:/Users/G/esp/esp-idf/components/mdns C:/Users/G/esp/esp-idf/components/mqtt C:/Users/G/esp/esp-idf/components/newlib C:/Users/G/esp/esp-idf/components/nghttp C:/Users/G/esp/esp-idf/components/nvs_flash C:/Users/G/esp/esp-idf/components/openssl C:/Users/G/esp/esp-idf/components/partition_table C:/Users/G/esp/esp-idf/components/protobuf-c C:/Users/G/esp/esp-idf/components/protocomm C:/Users/G/esp/esp-idf/components/pthread C:/Users/G/esp/esp-idf/components/sdmmc C:/Users/G/esp/esp-idf/components/soc C:/Users/G/esp/esp-idf/components/spi_flash C:/Users/G/esp/esp-idf/components/spiffs C:/Users/G/esp/esp-idf/components/tcp_transport C:/Users/G/esp/esp-idf/components/tcpip_adapter C:/Users/G/esp/esp-idf/components/ulp C:/Users/G/esp/esp-idf/components/unity C:/Users/G/esp/esp-idf/components/vfs C:/Users/G/esp/esp-idf/components/wear_levelling C:/Users/G/esp/esp-idf/components/wifi_provisioning C:/Users/G/esp/esp-idf/components/wpa_supplicant C:/Users/G/esp/esp-idf/components/xtensa
-- Configuring done
CMake Error at C:/Users/G/esp/esp-idf/tools/cmake/component.cmake:445 (add_library):
  Cannot find source file:

    C:/Users/G/esp/esp32_mouse_keyboard/main/ble_hid/esp_hidd_prf_api.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx
Call Stack (most recent call first):
  C:/Users/G/esp/esp-idf/tools/cmake/component.cmake:508 (idf_component_register)
  main/ble_hid/CMakeLists.txt:7 (register_component)

CMake Error at C:/Users/G/esp/esp-idf/tools/cmake/component.cmake:445 (add_library):
  No SOURCES given to target: __idf_ble_hid
Call Stack (most recent call first):
  C:/Users/G/esp/esp-idf/tools/cmake/component.cmake:508 (idf_component_register)
  main/ble_hid/CMakeLists.txt:7 (register_component)

-- Build files have been written to: C:/Users/G/esp/esp32_mouse_keyboard/build
cmake failed with exit code 1`

i am using latest esp tool for windows and esp-idf v4 release

n1rml, 2020-03-05:
I am not able to identify what the problem is, as the ble_hid folder is included in the components section

`Component paths: C:/Users/G/esp/esp-idf/components/app_trace C:/Users/G/esp/esp-idf/components/app_update C:/Users/G/esp/esp-idf/components/asio C:/Users/G/esp/esp32_mouse_keyboard/main/ble_hid C:/Users/G/esp/esp-idf/components/bootloader C:/Users/G/esp/esp-idf/components/bootloader_support C:/Users/G/esp/esp-idf/components/bt C:/Users/G/esp/esp-idf/components/coap C:/Users/G/esp/esp-idf/components/console C:/Users/G/esp/esp-idf/components/cxx C:/Users/G/esp/esp-idf/components/driver C:/Users/G/esp/esp-idf/components/efuse C:/Users/G/esp/esp-idf/components/esp-tls C:/Users/G/esp/esp-idf/components/esp32 C:/Users/G/esp/esp-idf/components/esp_adc_cal C:/Users/G/esp/esp-idf/components/esp_common C:/Users/G/esp/esp-idf/components/esp_eth C:/Users/G/esp/esp-idf/components/esp_event C:/Users/G/esp/esp-idf/components/esp_gdbstub C:/Users/G/esp/esp-idf/components/esp_http_client C:/Users/G/esp/esp-idf/components/esp_http_server C:/Users/G/esp/esp-idf/components/esp_https_ota C:/Users/G/esp/esp-idf/components/esp_https_server C:/Users/G/esp/esp-idf/components/esp_local_ctrl C:/Users/G/esp/esp-idf/components/esp_ringbuf C:/Users/G/esp/esp-idf/components/esp_rom C:/Users/G/esp/esp-idf/components/esp_websocket_client C:/Users/G/esp/esp-idf/components/esp_wifi C:/Users/G/esp/esp-idf/components/espcoredump C:/Users/G/esp/esp-idf/components/esptool_py C:/Users/G/esp/esp-idf/components/expat C:/Users/G/esp/esp-idf/components/fatfs C:/Users/G/esp/esp-idf/components/freemodbus C:/Users/G/esp/esp-idf/components/freertos C:/Users/G/esp/esp-idf/components/heap C:/Users/G/esp/esp-idf/components/idf_test C:/Users/G/esp/esp-idf/components/jsmn C:/Users/G/esp/esp-idf/components/json C:/Users/G/esp/esp-idf/components/libsodium C:/Users/G/esp/esp-idf/components/log C:/Users/G/esp/esp-idf/components/lwip C:/Users/G/esp/esp-idf/components/mbedtls C:/Users/G/esp/esp-idf/components/mdns C:/Users/G/esp/esp-idf/components/mqtt C:/Users/G/esp/esp-idf/components/newlib C:/Users/G/esp/esp-idf/components/nghttp C:/Users/G/esp/esp-idf/components/nvs_flash C:/Users/G/esp/esp-idf/components/openssl C:/Users/G/esp/esp-idf/components/partition_table C:/Users/G/esp/esp-idf/components/protobuf-c C:/Users/G/esp/esp-idf/components/protocomm C:/Users/G/esp/esp-idf/components/pthread C:/Users/G/esp/esp-idf/components/sdmmc C:/Users/G/esp/esp-idf/components/soc C:/Users/G/esp/esp-idf/components/spi_flash C:/Users/G/esp/esp-idf/components/spiffs C:/Users/G/esp/esp-idf/components/tcp_transport C:/Users/G/esp/esp-idf/components/tcpip_adapter C:/Users/G/esp/esp-idf/components/ulp C:/Users/G/esp/esp-idf/components/unity C:/Users/G/esp/esp-idf/components/vfs C:/Users/G/esp/esp-idf/components/wear_levelling C:/Users/G/esp/esp-idf/components/wifi_provisioning C:/Users/G/esp/esp-idf/components/wpa_supplicant C:/Users/G/esp/esp-idf/components/xtensa
-- Configuring done`

these are the errors

`CMake Error at C:/Users/G/esp/esp-idf/tools/cmake/component.cmake:445 (add_library):
  Cannot find source file:
   C:/Users/G/esp/esp32_mouse_keyboard/main/ble_hid/esp_hidd_prf_api.c
  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx
Call Stack (most recent call first):
  C:/Users/G/esp/esp-idf/tools/cmake/component.cmake:508 (idf_component_register)
  main/ble_hid/CMakeLists.txt:7 (register_component)
CMake Error at C:/Users/G/esp/esp-idf/tools/cmake/component.cmake:445 (add_library):
  No SOURCES given to target: __idf_ble_hid
Call Stack (most recent call first):
  C:/Users/G/esp/esp-idf/tools/cmake/component.cmake:508 (idf_component_register)
  main/ble_hid/CMakeLists.txt:7 (register_component)
-- Build files have been written to: C:/Users/G/esp/esp32_mouse_keyboard/build
cmake failed with exit code 1`

i am using latest esp tool for windows and esp-idf v4 release

n1rml, 2020-03-05:
I am not able to identify what the problem is, as the ble_hid folder is included in the components section

```
Component paths: C:/Users/G/esp/esp-idf/components/app_trace C:/Users/G/esp/esp-idf/components/app_update C:/Users/G/esp/esp-idf/components/asio C:/Users/G/esp/esp32_mouse_keyboard/main/ble_hid ...
```

these are the errors

```
CMake Error at C:/Users/G/esp/esp-idf/tools/cmake/component.cmake:445 (add_library):
Cannot find source file:
C:/Users/G/esp/esp32_mouse_keyboard/main/ble_hid/esp_hidd_prf_api.c
Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm.hpp .hxx .in .txx
Call Stack (most recent call first):
C:/Users/G/esp/esp-idf/tools/cmake/component.cmake:508 (idf_component_register)
main/ble_hid/CMakeLists.txt:7 (register_component)

CMake Error at C:/Users/G/esp/esp-idf/tools/cmake/component.cmake:445 (add_library):
No SOURCES given to target: __idf_ble_hid
Call Stack (most recent call first):
C:/Users/G/esp/esp-idf/tools/cmake/component.cmake:508 (idf_component_register)
main/ble_hid/CMakeLists.txt:7 (register_component)

-- Build files have been written to: C:/Users/G/esp/esp32_mouse_keyboard/build
cmake failed with exit code 1
```

i am using latest esp tool for windows(v2.2) and esp-idf v4 release

benjaminaigner, 2020-03-05:
Dear @n1rml 
as we do not have any CMake files in our project, I think you wrote one for yourself?
If possible, please provide the CMakeLists.txt file and I can have a look.

Then we could integrate them into the project.

Greetings

n1rml, 2020-03-05:
i dont know anything about cmake files so i just copied one from an example in edp-idf.
i looked at many examples and all of the CMakeLists.txt files were similar and didn't contain much information, just the project name
heres the file i'm using
[CMakeLists.txt](https://github.com/asterics/esp32_mouse_keyboard/files/4292944/CMakeLists.txt)

benjaminaigner, 2020-03-05:
Your example is basically correct, but there are additional files needed for each subfolder ("component" in esp-idf terms).

I've added working CMakeFiles. They were generated by the converter script: https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/build-system.html#automatic-conversion-tool

If there are any problems, feel free to re-open this issue.

Greetings

closed 2020-03-05 by benjaminaigner

n1rml, 2020-03-05:
Thank you very much