asterics/esp32_mouse_keyboard #33: MacOS Catalina does not reconnect

seanuleh, 2020-05-30:
I'm having issues getting reliable connections to my MacBook Pro. I can get the ESP to connect once but when the ESP is restarted or the device is disconnected, then it will not connect again. I can only get it to pair again by restarting my MacBook. Any ideas on how to tackle this?

seanuleh, 2020-05-31:
Unfortunately, this seems to be a MacOS issue so I am closing this ticket.

benjaminaigner, 2020-06-02:
May I ask whats the macOS issue? And maybe how to fix it :-)?

seanuleh, 2020-06-02:
Unfortunately, I am unsure what the issue is. But browsing the internet I found many threads with people having similar issues, there are plenty of example posts about this on reddit. Most the other complaints seem to be with bluetooth audio devices, but they exhibit the same symptoms. Unfortunately, there is no fix other than to rollback from Catalina.

I tested this sketch on another macbook with Mojave and can confirm that it works great, device reconnects as expected.

Unfortunately my macbook is a work device and I can't roll back, so I have this silly work around. Everytime I need to reset or disconnect the board from my macbook, i kill the bluetooth daemon. It comes up `blueetoothd` so you can just run:

```
ps -ef | grep -i blue
kill -9 <pid>
```

When the bluetooth daemon resets I am able to connect to the board. Occasionally, after a reset of the daemon, the first connection fails and I see the following logs in the serial monitor:
```
␛[0;32mI (23983) SEAN-GAP-EVENT: ESP_GAP_BLE_AUTH_CMPL_EVT␛[0m
␛[0;32mI (23983) hal_ble: remote BD_ADDR: 88e9fe7a67a0␛[0m
␛[0;32mI (23983) hal_ble: address type = 0␛[0m
␛[0;32mI (23983) hal_ble: pair status = fail␛[0m
␛[0;31mE (23993) BT_BTM: BTM_GetSecurityFlags false
␛[0m
␛[0;32mI (23993) hal_ble: fail reason = 0x55␛[0m
␛[0;31mE (24013) BT_BTM: BTM_GetSecurityFlags false
␛[0m
␛[0;31mE (24033) BT_BTM: BTM_GetSecurityFlags false
␛[0m
␛[0;31mE (24053) BT_BTM: BTM_GetSecurityFlags false
␛[0m
␛[0;31mE (24123) BT_BTM: BTM_GetSecurityFlags false
...
```

But after this, I can attempt to connect again from my mac and then it works. It would be great if you have time to try and help me get to the bottom of this :) But I am pretty sure the issue has nothing to do with your library. Note that other BLE projects/libraries I have tried on the esp32 all fail with the same patterns; even a sketch I tried on the Arduino library did the same thing.

rampadc, 2020-07-29:
Hey @benjaminaigner, I got the project to bond successfully with macOS Catalina 10.15.6 by making all read and write characteristics in `hid_device_le_prf.c` encrypted with MITM, and changed the `auth_req` as well.

`esp_ble_auth_req_t auth_req = ESP_LE_AUTH_REQ_SC_MITM_BOND;`

Enabling SMP tracing (`CONFIG_SMP_TRACE_LEVEL_DEBUG=y`), I recorded a log of the transactions at [https://gist.github.com/rampadc/e20eedeb66db39a24a00e78d7e59d8ba](https://gist.github.com/rampadc/e20eedeb66db39a24a00e78d7e59d8ba).

- Line [1](https://gist.github.com/rampadc/e20eedeb66db39a24a00e78d7e59d8ba#file-macos_bonded_successful_1-L1) is when the device first boot up, idling, advertising and waiting for a connection.
- Line [6](https://gist.github.com/rampadc/e20eedeb66db39a24a00e78d7e59d8ba#file-macos_bonded_successful_1-L6) is when I click "Connect" on the Mac.
- Line [172](https://gist.github.com/rampadc/e20eedeb66db39a24a00e78d7e59d8ba#file-macos_bonded_successful_1-L172) macOS is asking for a passkey. I changed `iocap` to `ESP_IO_CAP_IN` as my application requires it.
- Line [180](https://gist.github.com/rampadc/e20eedeb66db39a24a00e78d7e59d8ba#file-macos_bonded_successful_1-L180) sent a reply with the passkey back to macOS
- On the mac, I had to click "Enter". In both of my testing trials, I found SMP will pause at line [244](https://gist.github.com/rampadc/e20eedeb66db39a24a00e78d7e59d8ba#file-macos_bonded_successful_1-L244)
- After clicking "Enter" on the Mac, it keeps on going till the end of the trace file and bonded successfully.

Subsequent device resets auto paired as the device is bonded 👍 as shown in this [trace](https://gist.github.com/rampadc/9d213a1f2cd44c9f9474d1f16470ec49).

I changed the code a fair bit by adding passkey reply and I'm not sure how if MITM and ENC would work if you don't pair with no IO. What do you think? Worth a pull request? My modified code is at [https://github.com/rampadc/esp32_mouse_keyboard](https://github.com/rampadc/esp32_mouse_keyboard).

rampadc, 2020-07-29:
Also had to change the `GATTS_TAG` in `config.h` to something shorter. With the full `esp32_mouse_keyboard` string, I couldn't get the device to show up in macOS' Bluetooth list of devices.

benjaminaigner, 2020-07-29:
Dear @rampadc 
thank you very much for the effort to investigate this issue!
it's really nice, especially because you implemented $PK command as well :-).
I will test your branch with all available devices, if it works on all platforms, I will integrate it via a PR.
(should happen next week)

benjaminaigner, 2020-08-06:
@rampadc 

I've got a Catalina setup now, I can confirm problems with re-connecting.
First time connecting works well, if macOS is restarted, it works too (also if BT is switched off/on).
But: as soon as the ESP32 restarts, the connection does not work.

I will investigate:
* Problems with the GATT identifier
* If your changes work on all other platforms

benjaminaigner, 2020-08-06:
@rampadc 
I've tested your changes with most of my devices (or my employers devices :-)),
and your changes work well.
Opened a PR #40 
I've added a few suggestions there, if merged we can close this issue.

rampadc, 2020-08-06:
Hey @benjaminaigner the PR looks good. I'm all for a merge. I'll look into the esp32 restart issue in the next 2 weeks.

benjaminaigner, 2020-08-10:
I've changed the PR & added my suggestions in a new PR #42 
If anybody can confirm it works on her/his system (it does on my macOS system), I will merge.

seanuleh, 2020-08-15:
Thanks for your efforts on this both of you. I have tested #42 on my macOS system and can confirm that it is working and reconnecting when the ESP32 resets. 

There is just one weird issue I have found, but I don't think its a big deal. When you remove the device from macOS, reconnecting the device fails. You have to remove it again and then retry connecting one more time for it to connect again. 
Here are logs for this:

Start with device connected

> BT -> Remove

```
I (100218) HID_DEMO: ESP_HIDD_EVENT_BLE_DISCONNECT
```

> BT -> Connect: 

```
I (119538) HID_LE_PRF: HID connection establish, conn_id = 0
I (119548) HID_DEMO: ESP_HIDD_EVENT_BLE_CONNECT
I (119548) HID_LE_PRF: GATT EVT 4
E (120218) BT_SMP: Value for numeric comparison = 661199
I (120218) HID_DEMO: Requesting confirmation, we will send YES...
I (120218) HID_DEMO: Requester's BD_ADDR: 88e9fe7a67a0
E (121198) BT_GATT: GATT_INSUF_ENCRYPTION

I (130578) CONSOLE_UART: Not connected, ignoring 'a'
I (130838) CONSOLE_UART: Not connected, ignoring 'a'
I (131058) CONSOLE_UART: Not connected, ignoring 'a'
E (140008) BT_BTM: Device not found
```
Says connected in macOS, but doesn't really work

> BT -> Remove:
```
I (140018) HID_DEMO: ESP_HIDD_EVENT_BLE_DISCONNECT
I (140038) HID_DEMO: remote BD_ADDR: 88e9fe7a67a0
I (140038) HID_DEMO: address type = 0
I (140038) HID_DEMO: pair status = fail
E (140038) HID_DEMO: fail reason = 0x66
```

> BT -> Connect
```
I (192018) HID_LE_PRF: HID connection establish, conn_id = 0
I (192018) HID_DEMO: ESP_HIDD_EVENT_BLE_CONNECT
I (192018) HID_LE_PRF: GATT EVT 4
E (192508) BT_SMP: Value for numeric comparison = 968047
I (192508) HID_DEMO: Requesting confirmation, we will send YES...
I (192508) HID_DEMO: Requester's BD_ADDR: 88e9fe7a67a0
E (193678) BT_GATT: GATT_INSUF_AUTHENTICATION

W (193898) BT_SMP: FOR LE SC LTK IS USED INSTEAD OF STK
I (194628) HID_LE_PRF: GATT EVT 1
I (194628) HID_LE_PRF: GATT EVT 1
I (194628) HID_LE_PRF: GATT EVT 1
I (194638) HID_DEMO: remote BD_ADDR: 88e9fe7a67a0
I (194638) HID_DEMO: address type = 0
I (194638) HID_DEMO: pair status = success
```

benjaminaigner, 2020-08-15:
@seanuleh 
Thanks for this good news. Do you tested it with Catalina?
Because it seems we have to decide, either Catalina or Mojave (see #41 )

seanuleh, 2020-08-15:
Yes, tested with Catalina, but some weird stuff is happening now :( Might need someone else to test. I cannot get it to reconnect now even after multiple restarts. 

I've actually been using this project to build a keyboard: https://github.com/Galzai/MK32 last night I received an update to Version 10.15.6 and today when I tested again with this project it was also reconnecting fine. So I believe whatever bug was causing this issue is fixed. 

However, I tried to prove this by loading up your firmware from the master branch and now I just can't get my ESP32 to reconnect to my macbook even after multiple restarts and switching back to the fixCatalina branch. The MK32 project still works however and actually now stays reconnected through ESP32 restarts... my environment seems a bit cooked right now, perhaps you can retest without these fixes?