asterics/esp32_mouse_keyboard #34: iOS: Assistive Touch & Switch interface

benjaminaigner, 2020-06-23:
After a very long try&error phase with this undocumented stuff from iOS,
I can finally announce that this repository works with iOS in assistive touch mode (mouse cursor) AND switch interface (button controlled scanning).

For future references (or maybe the esp32_snippets repository), following issues were resolved:

* If the HID report map (and the corresponding attribute table) lists the mouse first, everything works fine. But iOS does NOT recognize any keyboard input as assistive switch input.
* If the report map is built up: keyboard - keyboard LEDs - consumer control - mouse; everything works exactly the same, but iOS recognizes key presses as assistive switch.
* The slave connection interval must have a value higher than 11.25ms (documented), we changed it to 12.5ms. This was no issue for normal keyboard operation & assistive touch mouse control.
* The MTU does not have any effect on the connection.

I did not find any of this behaviour documented, if somebody can show me an official guideline for this, I'm fine. Otherwise, this is another example of undocumented and senseless behaviour of Apple devices :-1:

benjaminaigner, 2020-06-29:
Closed with #35

closed 2020-06-29 by benjaminaigner

curiousmuch, 2021-12-13:
Can I buy you a beer for this?

benjaminaigner, 2021-12-13:
Dear @curiousmuch 
THX, but no worries: there is enough beer in Austria :+1: