asterics/esp32_mouse_keyboard #38: Enable notifications for LED reports

rampadc, 2020-07-17:
Can you enable notifications for LED reports? At the moment, when I turn on/off Capslock, Numlock and Scrolllock, the ESP32 isn't notifying me.

benjaminaigner, 2020-07-17:
Dear @rampadc ,

yes, I will enable it.
Due to our primary use case for this software, it was no issue until now, but if you request it, no problem!

benjaminaigner, 2020-07-20:
@rampadc 
It's working now, closed with 24b0f8d .

I will close this issue, if you have any troubles feel free to re-open :-).

Usage: I introduced a new HID event (`ESP_HIDD_EVENT_BLE_LED_OUT_WRITE_EVT`) which can be received
in the HID callback (`hidd_event_callback`) in ble_hidd_demo_main.c

THX for the hint!

closed 2020-07-20 by benjaminaigner