asterics/esp32_mouse_keyboard #41: GATT_INSUF_AUTHENTICATION problem

shownb, 2020-08-08:
at first I can connect to the MacBook without any problem.
after I deep_sleep the esp32 then shutdown the MacBook.
the active the esp32,the turn on the MacBook.
esp32 shows that
```
I (636) BTDM_INIT: BT controller compile version [d9cc6b1]
I (636) system_api: Base MAC address is not set
I (646) system_api: read default base MAC address from EFUSE
I (666) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 1
I (1086) HID_LE_PRF: esp_hidd_prf_cb_hdl(), start added the hid service to the stack database. incl_handle = 40
I (1086) HID_LE_PRF: hid svc handle = 2d
I (1096) HID_LE_PRF: GATT EVT 12
I (1096) HIDD: MAIN finished...
I (1096) Keyboard task: initializezd
I (1106) Sleep: initializezd
I (1116) HID_LE_PRF: GATT EVT 12
I (1226) HID_LE_PRF: HID connection establish, conn_id = 0
I (1226) hal_ble: ESP_HIDD_EVENT_BLE_CONNECT

W (34986) BT_BTM: btm_sec_clr_temp_auth_service() - no dev CB

I (34986) hal_ble: remote BD_ADDR: aaaaaaaaaaaa
I (34986) hal_ble: address type = 0
I (34986) hal_ble: pair status = fail
E (34996) hal_ble: fail reason = 0x63
I (34996) hal_ble: ESP_HIDD_EVENT_BLE_DISCONNECT
E (35016) BT_BTM: Device not found

I (66716) HID_LE_PRF: HID connection establish, conn_id = 0
I (66716) hal_ble: ESP_HIDD_EVENT_BLE_CONNECT
I (66726) HID_LE_PRF: GATT EVT 4
I (66846) HID_LE_PRF: GATT EVT 1
E (66966) BT_GATT: gatts_write_attr_perm_check - GATT_INSUF_AUTHENTICATION
I (97476) hal_ble: remote BD_ADDR: aaaaaaaaaaaa
I (97476) hal_ble: address type = 0
I (97476) hal_ble: pair status = fail
E (97476) hal_ble: fail reason = 0x63
```

and what is the problem?and how can I improve it. thanks for the great job.

benjaminaigner, 2020-08-10:
With the help of https://github.com/rampadc we added compatibility with new macOS versions.

Please provide following information:
* which macOS Version

If possible, please try our bugfix:
https://github.com/asterics/esp32_mouse_keyboard/tree/fixCatalina
(either you download the code with the green button, or `git checkout fixCatalina`)

If you can confirm that it works, it would really help us to merge this bugfix without headache (more tested devices :+1)

THX

shownb, 2020-08-10:
no luck
```
I (551) HID_DEMO: pairing enabled by default
I (571) BTDM_INIT: BT controller compile version [219866f]
I (571) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (591) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 1
I (941) MAIN: loading configuration from NVS
I (941) MAIN: bt device name is: HID
I (951) MAIN: error reading NVS - locale, setting to US_INTERNATIONAL
I (961) UART: console UART processing task started
I (961) EXT_UART: external UART processing task started
I (961) HID_LE_PRF: esp_hidd_prf_cb_hdl(), start added the hid service to the stack database. incl_handle = 40
I (981) HID_LE_PRF: hid svc handle = 2d
I (981) HID_LE_PRF: GATT EVT 12
I (1001) HID_LE_PRF: GATT EVT 12
I (1081) HID_LE_PRF: HID connection establish, conn_id = 0
I (1081) HID_DEMO: ESP_HIDD_EVENT_BLE_CONNECT
E (6971) BT_BTM: Device not found

I (6971) HID_DEMO: ESP_HIDD_EVENT_BLE_DISCONNECT
I (7181) HID_DEMO: remote BD_ADDR: 98460a8ee071
I (7181) HID_DEMO: address type = 0
I (7181) HID_DEMO: pair status = fail
E (7191) HID_DEMO: fail reason = 0x66
I (24461) HID_LE_PRF: HID connection establish, conn_id = 0
I (24461) HID_DEMO: ESP_HIDD_EVENT_BLE_CONNECT
I (24481) HID_LE_PRF: GATT EVT 4
E (24551) BT_GATT: GATT_INSUF_AUTHENTICATION

E (24711) BT_GATT: gatts_write_attr_perm_check - GATT_INSUF_AUTHENTICATION
E (24801) BT_GATT: gatts_write_attr_perm_check - GATT_INSUF_AUTHENTICATION
E (24911) BT_GATT: gatts_write_attr_perm_check - GATT_INSUF_AUTHENTICATION
E (25001) BT_GATT: gatts_write_attr_perm_check - GATT_INSUF_AUTHENTICATION
I (38831) CONSOLE_UART: received: 27, no HID action
I (38831) CONSOLE_UART: received: 91, no HID action
I (38831) CONSOLE_UART: received: 65, no HID action
I (54581) HID_DEMO: remote BD_ADDR: 98460a8ee071
I (54581) HID_DEMO: address type = 0
I (54581) HID_DEMO: pair status = fail
E (54581) HID_DEMO: fail reason = 0x63
W (57471) BT_BTM: btm_sec_clr_temp_auth_service() - no dev CB

I (57471) HID_DEMO: ESP_HIDD_EVENT_BLE_DISCONNECT
E (57501) BT_BTM: Device not found
```

benjaminaigner, 2020-08-10:
I think I found the problem. Please give it another try.
(`git pull && git checkout fixCatalina && git pull`)

Just to be sure: please wipe the ESP32 completely before testing (clearing any remaining STK/LTK BLE keys).

THX!

shownb, 2020-08-10:
funktioniert im Moment sehr gut.

I will try it several times

benjaminaigner, 2020-08-10:
@shownb 
Danke :+1: 
It worked flawlessly now, no idea whats the difference :-). 

I used a MB Pro 2012, Mojave 10.14.6.
I tested this branch with a VBox VM as well (some kind of hackintosh ^^), where I have Mojave 10.14.2 & Catalina 10.15.6.

I'v noticed, that macOS in general does not really like en-/disabling Bluetooth.

shownb, 2020-08-10:
@benjaminaigner 
It's pity to say that, the bad thing happen again.I runned `make erase_flash` and got the same problem.

benjaminaigner, 2020-08-10:
As I can see, you have the same troubles with the Espressif example
(https://github.com/espressif/esp-idf/issues/5711).

Do you have any additional hint what leads to this problem?
Did you reset the ESP32, does it woke up from deep sleep or was a restart of the macOS machine the reason this time?

benjaminaigner, 2020-08-10:
My one (and unfortunately currently only) hint for you:
* Please try to set 
`esp_ble_auth_req_t auth_req = ESP_LE_AUTH_REQ_SC_MITM_BOND;`
to different values
(they are defined in `components/bt/host/bluedroid/api/include/api/esp_gap_ble_api.h`).

And if none of these work as expected, I fear we need to switch to a passkey input.

For the passkey input, please replace:
`esp_ble_io_cap_t iocap = ESP_IO_CAP_IO;`
with
`esp_ble_io_cap_t iocap = ESP_IO_CAP_IN;`
and enter via the console the shown pin on the host device via `$PK <passkey>`

shownb, 2020-08-11:
I will try it and reply here.but need some time.because a lot of erase_flash & make. thank you for the great work.

benjaminaigner, 2020-08-11:
@shownb 
Take your time, better we merge a well functioning PR :+1: 
THX for testing!!!

shownb, 2020-09-02:
sorry for so late.
but it seems like that I bring the good news.
I tried it with the example that in this url https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/ble/gatt_security_server/main/example_ble_sec_gatts_demo.c
```c
    /* set the security iocap & auth_req & key size & init key response key parameters to the stack*/
    esp_ble_auth_req_t auth_req = ESP_LE_AUTH_REQ_SC_MITM_BOND;     //bonding with peer device after authentication
    esp_ble_io_cap_t iocap = ESP_IO_CAP_NONE;           //set the IO capability to No output No input
    uint8_t key_size = 16;      //the key size should be 7~16 bytes
    uint8_t init_key = ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK;
    uint8_t rsp_key = ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK;
    //set static passkey
    uint32_t passkey = 123456;
    uint8_t auth_option = ESP_BLE_ONLY_ACCEPT_SPECIFIED_AUTH_DISABLE;
    uint8_t oob_support = ESP_BLE_OOB_DISABLE;
    esp_ble_gap_set_security_param(ESP_BLE_SM_SET_STATIC_PASSKEY, &passkey, sizeof(uint32_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_AUTHEN_REQ_MODE, &auth_req, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_IOCAP_MODE, &iocap, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_MAX_KEY_SIZE, &key_size, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_ONLY_ACCEPT_SPECIFIED_SEC_AUTH, &auth_option, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_OOB_SUPPORT, &oob_support, sizeof(uint8_t));
```

benjaminaigner, 2020-09-02:
:+1: :+1: :+1: 

Thank you very much for the hint! I would like to test it right now, but I've no access to all my devices until Monday.
I will report back with results and if it is working, we can close this issue and hopefully #33  too.

benjaminaigner, 2020-11-13:
@shownb 

I've tested macOS Catalina (v10.15.7) with the current master and I had no problem with reconnecting.
So I'm not sure if we should apply your patch.

Would it be possible for you to test our master with your device again?

Greetings

shownb, 2020-12-01:
@benjaminaigner 
I'm sorry for the late reply. I'm sorry I don't have time to test it at the moment, but up until the last time I tested it, everything was actually ok. so I think we found the workaround, may be you can close the issue!
finnally wish you Merry Christmas and staying healthy.

benjaminaigner, 2020-12-01:
@shownb 
THX for the feedback, I will close this issue now. Feel free to reopen if any problem occurs.

Thanks for the wishes, I wish you and your beloved also nice holidays and merry christmas!

closed 2020-12-01 by benjaminaigner