asterics/esp32_mouse_keyboard #43: Switch between devices/hosts

Juno72, 2020-09-21:
Getting an ESP32 and want to wire my broken keyboard.
Having a MX vertical switching between hosts/computers is ideal.
Figured out the BluetoothDeviceAddress in the windows properties is different when connecting to another id of the mouse: fc5cabc8e5a2 vs fc5cabc8e5a4. ProductId (20 B0), Vendor(6D 04), VenderSource (2) remains equal.

So using above switching from host would/should require disconnect current connection, set another MAC address and setup (new) connection.

Alternatively one can setup/allow multiple connection at the same time, and send keys to only one (or more) selected connection(s).
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gattc_multi_connect
 

Have to wait for my board and setup a build environment using my limited time and other projects, but for now maybe ideas for others.

Juno72, 2020-09-21:
The MX vertical denies paring a server/Windows10 twice, probably the host mac address is verified in the other 2 connections/mac address. The MX vertical can connect 3 hosts.

benjaminaigner, 2020-11-13:
Dear @Juno72 ,

as mentioned in #36 we have troubles with the ESP32 whitelisting.
So a device selection based on pairings could not be done without problems (currently).

Thank you for the link to the example of multiple GATT connections.
I can't tell you any timeline to implement this, because it is not our top priority.

I'm open to a PR if somebody else has the time for this feature :-)

Greetings

benjaminaigner, 2020-12-01:
@Juno72 

I had enough time now on the Magenta hotline to implement your request :-).

With 4df05f4 you can switch between different connected devices via:

`$GC` Get all active connections, returns the BT MAC addresses
`$SW <macaddr>` Select this MAC address to send the HID data to.

example:
`$SW 47A8DBEA5915`

__Note:__ If you enter a command on the ESP32 debug output, the movement with w/a/s/d is disabled. As we use this project as addon board for the FLipMouse, I'm usually sending the
commands via the external UART interface.

benjaminaigner, 2020-12-01:
Feel free to reopen if you experience any problems.

closed 2020-12-01 by benjaminaigner