asterics/esp32_mouse_keyboard #45: Can't pair with Apple TV (4th generation) or iPhone 11 Pro Max -- esp32 doesn't even appear in list

sofakng, 2020-11-12:
I'm trying to connect to my Apple TV (4th generation) but when I open the blueooth menu, the esp32 doesn't appear as a device I can pair.  The same problem exists on my iPhone 11 Pro Max.

However, I do see the esp32 device on my Windows 10 machine (under bluetooth devices) so I believe the esp32 itself is working properly...

benjaminaigner, 2020-11-13:
Dear @sofakng 

could you please tell us the iOS version which is used by your devices?
As you can see (#33 #41) we have problems with different apple based devices.
I'm trying to test the firmware against the current version of iOS; but as it seems we cannot cover every case :-).

Greetings

PS: are you related with the sofaking account :-)?

benjaminaigner, 2022-09-05:
Closing because of inactivity.

closed 2022-09-05 by benjaminaigner