asterics/esp32_mouse_keyboard #51: How do you send key commands

jonathanrobichaud4, 2021-07-16:
I can send regular commands to rename the esp but I cannot send the hex code. GPIO17/16 doesn't work and I also hooked directly up with the rx/tx pins. what could I be missing

benjaminaigner, 2021-07-22:
Dear @jonathanrobichaud4 

sry for the long answer delay, holiday time here :-).
We can support you as much as possible, but we do need some more infos.

* You can change the name of the ESP32 via the `$NAME xxx` command?
* You can pair with the ESP32?
* How do you send the mouse/keyboard commands (other uC; terminal program;....)?

If the commands are processed (`$NAME`, `$ID`,...), the serial interface works as expected,
so should the HID commands.
Sending HID commands is done in this format:
https://github.com/asterics/esp32_mouse_keyboard#hid-input
But not in ASCII command format, but binary.

jonathanrobichaud4, 2021-07-22:
The commands do work and I can pair with the esp just fine. Right now I am using a program named serial which is for Mac devices. I put it into the hex mode and tried to use it with the start bit and everything. "fd 00 00 73" kind of like that. I tried with the 0x in front of each as well. If you need anything else please let me know! I am writing this on my phone so it's a bit sparse sorry.

benjaminaigner, 2021-07-22:
Dear @jonathanrobichaud4 

OK, thx for the infos. What about:
* What is the BT central device (PC/Notebook or mobile; which host OS)
* In many serial terminal programs, a `\r` or `\n` are sent as well. This should be switched off in your program
* What serial interface of the ESP32 is used (UART0 with the debug output or UART2)?

jonathanrobichaud4, 2021-07-22:
It's an android device and I'm not sure what uart is in use. And I will check later today if my program sends the \r or \n.

jonathanrobichaud4, 2021-07-22:
what program do you recommend for communication? Putty or some other windows specific one? I have a windows pc I can fire up as well. How can I tell what uart interface I am using. Right now I'm communicating through usb but can do the commands like $NAME as well as the demo ones

benjaminaigner, 2021-07-23:
Hi,
OK, Android devices should work without problem.
Regarding the UART: I think this is the source of your problem:
* We have 2 functions for parsing UART data: https://github.com/asterics/esp32_mouse_keyboard/blob/77103b7c01d78e6d53b74c861d9d6b035679ff45/main/ble_hidd_demo_main.c#L966 and https://github.com/asterics/esp32_mouse_keyboard/blob/77103b7c01d78e6d53b74c861d9d6b035679ff45/main/ble_hidd_demo_main.c#L1059
* `uart_console_task` processes the data received from the UART0, which is the programming and debugging UART (connected via USB; normally used with `idf.py flash/monitor`
* `uart_external_task` processes the data from the second UART (pin 16/17 for RX&TX)
* The console task does NOT implement the RAW HID commands. We just used this task for testing mouse/keyboard via the keys `w,a,s,d,c` for mouse and `q` for keyboard.

Unfortunately, we didn't need the RAW HID for the UART0 (console_task), so it is not implemented.

A fast solution would be:
https://github.com/asterics/esp32_mouse_keyboard/blob/77103b7c01d78e6d53b74c861d9d6b035679ff45/main/ble_hidd_demo_main.c#L1077
After this line, add `uart_parse_command(character,&commands);`
And break the while(1) loop directly afterwards.

jonathanrobichaud4, 2021-07-23:
I tried that out but it didn't help. when I tried sending commands I got a freertos error and the esp reset

jonathanrobichaud4, 2021-07-23:
it says :

`FreeRTOS: FreeRTOS Task "console" should not return, Aborting now!`

benjaminaigner, 2021-07-26:
1. Have you tried it without modification and sending w,s,a,d keystrokes to the ESP32? Does the mouse move accordingly?
2. Did you use `break;`, `return;` or `continue;` after uart_parse_command?

jonathanrobichaud4, 2021-07-31:
I tried a few different things but I just ended up using a different project that worked for what I needed it to do! Thanks for the help though!

closed 2021-07-31 by jonathanrobichaud4

benjaminaigner, 2021-08-02:
Dear @jonathanrobichaud4 

I'm glad you find something that worked for you!
Anyway it would be nice if you give our project a star :+1: 
Good luck with your project,

greetings