asterics/esp32_mouse_keyboard #53: help with HID braille for open source devices for people with disabilities

discapacidad5, 2021-12-06:
Hello, I am developing a device for people with disabilities. I need to use the HID braille protocol but I am not getting anything with HID for esp32, could you help me with a sample code of how I can implement the HID braille protocol?

here are the braille HID references

https://usb.org/sites/default/files/hutrr78_-_creation_of_a_braille_display_usage_page_0.pdf

https://github.com/nvaccess/nvda/blob/00cd67b737bc5b23a6f5e31cf28110b64ebf2fee/devDocs/hidBrailleTechnicalNotes.md

my repository is the following
https://github.com/brailletouch/Brailletouch

benjaminaigner, 2021-12-06:
Dear @discapacidad5 ,
very interesting approach with one real braille cell and the virtual cells!
Regarding the Bluetooth HID implementation:
* unfortunately, we don't need a Braille implementation in this repository (maybe my colleagues will find a use case, but I can't). Nevertheless, if you create a new repository for the ESP32 firmware (I would recommend to split the firmware & case designs) I can contribute.
* Thanks for the USB reference document, I think there is everything you need:

To change the HID functionality you need following steps (mostly in the file _hid_device_le_prf.c_):
1. Change the HID descriptor: I would simply use the example from the referenced document. It seems to include left/right control buttons, braille input and some buttons (look for _collections_ which end with an _input_ tag)
2. Change the report structure: On the first read, there seems to be only one input and one output report (no report IDs). Combining all report size/report count tags in the HID descriptor you will get the bit/byte positions in your report where data should be sent / read.

Hope I could help, I like your projects!

Greetings

discapacidad5, 2021-12-06:
Thank you for answering. I tell you that I do not speak English, I speak
Spanish and I am using a Google translator, it is probably easy for you to
understand all the documentation that I give you, but for me it is
unintelligible that is why I seek help from someone who does understand
you. And that I can make an example code from which I can guide myself
because there is a lot of documentation but there is no practical example
and when it comes to documentation in English when translating into Spanish
a lot of essence is lost besides I am a baby in programming I am in full
learning that makes me even more difficult

 So if you could make me an example code with that that you showed me in
your suggestion it would be very helpful for me because I don't know how
the HID code works and I don't really know how to do what you are
recommending.

benjaminaigner, 2021-12-06:
OK, I would get started with the Espressif example:
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/ble_hid_device_demo
There is everything you need to get started based on a mouse/keyboard project (exactly as we did here).
Try to compile and flash the example, to ensure a correct development setup.
(getting started for the ESP32: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html )

And then start to change the HID descriptor to the example from your document.

benjaminaigner, 2021-12-07:
@discapacidad5 

My colleague @ChrisVeigl remembered me of an existing repository where a student of us was building
a refreshable braille device:
https://github.com/ProjectKitchen/BrailleDisplay/

Maybe you can find some useful information for building your device.

benjaminaigner, 2022-09-05:
Closing as off-topic, maybe we can consider implementing HID braille in a separate repository

closed 2022-09-05 by benjaminaigner