asterics/esp32_mouse_keyboard #54: Compilation issue.

manimathma, 2021-12-13:
I am getting compilation issue, am i missing something basic.

ble_hidd_demo_main.c:340: undefined reference to `esp_ble_gap_start_advertising'

benjaminaigner, 2021-12-13:
Dear @manimathma ,

interesting. According to the esp-idf documentation, this function is included at least since v3:
https://docs.espressif.com/projects/esp-idf/en/release-v3.3/api-reference/bluetooth/esp_gap_ble.html

Could you please tell me what esp-idf version you are using?

Greetings

manimathma, 2021-12-13:
--- a/main/CMakeLists.txt
+++ b/main/CMakeLists.txt
@@ -2,6 +2,6 @@ idf_component_register(SRCS "ble_hidd_demo_main.c"
                             "esp_hidd_prf_api.c"
                             "hid_dev.c"
                             "hid_device_le_prf.c"
-                    INCLUDE_DIRS ".")
+                    INCLUDE_DIRS "." REQUIRES esp_hid PRIV_REQUIRES esp32 esp_wifi esp_https_server esp_eth mdns lwip fatfs openssl esp_https_ota esp_hid app_update) 

Able to resolve the issue with the above patch. 

Didn't get sufficient time to dive deep into why this diff is needed. 

I am using IDF 4.3 and i am building for C3F.

benjaminaigner, 2021-12-13:
@manimathma 
Thank you very much for the quick fix! I'm also not really sure why this is needed, especially because we
are not used _esp_hid_ in our project. But I think this includes the BLE API and that's what we need :+1:

I've added the patch and added the esp_websocket_client as well (we will need this for the I2S microphone).

closed 2021-12-13 by benjaminaigner