asterics/esp32_mouse_keyboard #57: Pairing problems on Win8.1

ChrisVeigl, 2022-01-23:
Using Firmware version 0.3.0, I cannot pair with a Lenovo Yoga Pro Laptop running Win8.1
(where pairing works with Win10/Linux PC and Android/Linage Smartphone)

benjaminaigner, 2022-01-25:
Hi,
it is hard to determine the cause of this error (because I don't have access to a Win8 installation).
If possible, you could flash this project on an ESP32 devboard with UART0->USB connection and post the logs here.

It would be a nice feature (especially for debugging enduser problems) to redirect the UART0 debug output temporarily to UART1 and therefore the host device (FM/FABI/FP) which redirects it to the host USB (could be visible in the WebGUI for copy&paste).
But after a quick research, I didn't find a suitable option to do this on-the-fly.

Maybe the faster option: you bring the notebook :-)

benjaminaigner, 2023-02-16:
I will close this issue now, as Windows 8.1 is not supported anymore (since 10.01.2023).

closed 2023-02-16 by benjaminaigner