asterics/esp32_mouse_keyboard #61: How to get or set mouse position?

m986883511, 2022-03-10:
I want to click on a specific place on the screen, is there a way to get the position of the mouse on the screen, or click on a certain place？

benjaminaigner, 2022-03-10:
Dear @m986883511 ,

unfortunately with the current firmware this is NOT possible without any workarounds.
You need an absolute mouse pointer for this, as it is implemented with e.g. Wacom Tablets.

There is a branch here with a test implementation:
https://github.com/asterics/esp32_mouse_keyboard/tree/absoluteMouse

You could try that.

Or if there is no other input device, you can try:
* Move the mouse cursor to the left upper corner by sending many commands with negative x & y numbers
* Move the mouse cursor to your designated position by counting up your sent values for x & y axis.

But beware: the values you send from the ESP32 are NOT pixels. The OS calculates with the mouse settings (acceleration, sensitivity, ...) a different value for the actual pixels, so this is try&error.

benjaminaigner, 2022-09-05:
@m986883511 
Closing this issue, if you have any update or even a pull request, feel free to reopen.

closed 2022-09-05 by benjaminaigner