asterics/esp32_mouse_keyboard #65: Not showinfg on HID device in Windows 10

lodogg, 2024-02-14:
"Good morning, I'm trying to use the code on an ESP32-S3. When I connect it to my Android phone, everything works correctly, but when I connect it to the PC, it doesn't work. In particular, I noticed that it doesn't appear among the HID devices (which instead happens with another similar example Arduino arduino-switcheroonie). Any ideas about this?"

benjaminaigner, 2024-02-14:
Dear @lodogg 
thx for this info.
Did you have a look into the Windows Device Manager?
Please tell me which build config you are using (whether with or without joystick/keyboard/mouse).

lodogg, 2024-02-14:
I'm using the last release 0.3.6
This is my actual Windows device manager:

benjaminaigner, 2024-02-14:
THX; I see. Please post the sdkconfig file here and I hope I have spare time and a ESP32-S3 to test it.

In the meantime: please try to compile it again without Joystick.

lodogg, 2024-02-14:
This is my sdkconfig. Tried without joystick with no success. If you need some help or test something let me know. 
[sdkconfig.txt](https://github.com/asterics/esp32_mouse_keyboard/files/14279765/sdkconfig.txt)

benjaminaigner, 2024-03-07:
Dear @lodogg ,

your sdkconfig is missing `CONFIG_MODULE_USEJOYSTICK`, so I guess it is not the current version of this repository.
I've tried it yesterday, without any problem (fresh clone; fresh build without changing anything) on a Windows 10 Pro.
If possible, could you provide the log output of the serial console?
Next thing I'll try is using the ESP32-S3 (I've used the classic one).

benjaminaigner, 2024-03-13:
Dear @lodogg ,
for today, I've got the firmware running on an ESP32-S3 (see #66 ).
It's a little bit funny: the bootloader is producing log messages, but as soon as we hit `I (0) cpu_start: Starting scheduler on APP CPU.`, there is no further output (also on the USB-CDC interface); but the device starts (I can pair, connect & reconnect).
If I have time, I will investigate further.

benjaminaigner, 2024-03-14:
@lodogg 

I've successfully built and tested the firmware on the ESP32-S3 as well, please find my sdkconfig attached, which worked for me (esp-idf version v5.1.1).
[sdkconfig.txt](https://github.com/asterics/esp32_mouse_keyboard/files/14604715/sdkconfig.txt)

closed 2024-03-14 by benjaminaigner